skuhl/RobotRun #5: Fix inverse kinematics

skuhl, 2016-05-03:
Scott will provide a quick tutorial after you feel like you have an idea for how code that controls the robot arm.

neoaero99, 2016-05-04:
Would fixing inverse kinematics make the robot more precise? I have noticed that the robot is not very precise in appoarching a point you stored for a program.

vtdrusch, 2016-05-04:
I think that those are probably unrelated issues, since returning to a
stored position should just be a matter of setting the joint positions to
be the stored values. If there's a difference between the values being
stored and the position of the robot, there's probably an issue with the
saving or loading of positional data.

skuhl, 2016-05-04:
Inverse kinematics tries to efficiently search for a set of joint angles which will get the end effector "close enough" to where it is supposed to be. The programmer gets to define "close enough" and making it be really, really close to accurate increases the computational cost. I know that they way that James implemented it is similar, you should be able to find the "close enough" hardcoded value and/or some kind of loop that keeps searching for a solution until the end effector is close enough.

Inverse kinematics doesn't just worry about how close it gets, it also typically want to minimize the amount of joint movement to move from one arm orientation to another (i.e., if you can get the end effector with a small movement of 1 joint or a large movement of 1 joint, the small movement is preferred, for example).

vtdrusch, 2016-05-10:
Made some headway here, IK algorithm with Jacobian transpose is working, albeit with noticeable frame rate drop when approaching the limit of the arm's range of motion. Still, the simple Jacobian algorithm used here seems to be more efficient and more accurate than the previous implementation, and eliminates the random, jerky rotation.

The efficiency and resolution of the algorithm are closely related to the values used to calculate the magnitude of each delta in the Jacobian array (J) and the value of the resulting angular change for each joint: in order for the algorithm to actually converge on the target point, the change in angles for each joint has to be sufficiently small for each cycle of the algorithm. Specifically, the deltas for each round of calculations should be significantly smaller than the total distance being moved (the algorithms "search" for the correct angles for the given position has to be fine grain enough to incorporate very small changes in end effector position). Currently, I'm calculating the EE deltas for J using a 1 degree offset (~0.0174 RAD) and a coefficient of about 1/1000th of a degree on the angles obtained from the dot product of  J and the EE target deltas. Reducing the angular offsets by such a degree helps to ensure that the algorithm doesn't "overshoot" the target position during its calculations, but increases the number of cycles required to obtain the correct offsets. My suspicion is that these two values can be optimized to improve the IK algorithm's efficiency.

I also have yet to implement the calculation for an alpha value, which should further improve my IK calculation times, but we're getting there.

neoaero99, 2016-05-10:
Alright, that sounds good. I am trying to implement the joint ranges for
program executation, but I am having little success with it.

vtdrusch, 2016-05-10:
Alpha value calculation has been implemented and, as Scott had suggested, does wonders for performance. The alpha value is essentially a smarter version of the coefficient on angular offset that I had mentioned previously, and automatically scales up or down to provide the best resolution possible for each IK calculation. World jogging is now very smooth and accurate (within 1 unit) for all reachable points up until the very edge of the arm's reach, and my next task will be looking into limiting the arm's motion and preventing movement requests to points outside of a certain range.

The next major addition will be including rotational IK. Currently, only translational jogging is possible (for either method of IK calculation), and the rotational value of the end effector is not preserved when doing so. Rotational values (WPR values) need to be locked when translating the end effector, and rotational jogging needs to be implemented in world coordinate mode as well.

neoaero99, 2016-05-11:
Yeah, I noticed that the rotational IK is not implemented, which is why I
could not accurately finished the button highlighting while jogging in
world coordinate mode.

vtdrusch, 2016-06-05:
Finally starting to see the light at the end of the tunnel in terms of completing the inverse kinematics package (which has really worked just fine for the past couple weeks, the issue was mostly with how to actually specify changes in orientation in a way that made sense for the algorithm; once you know where you're going, inverse kinematics can get you there). I'm happy to report that world frame rotational movement is now (mostly) working! I made a few changes to the way that quaternion rotations were calculated and a fixed a few bugs in my matrix/ quaternion/ Euler code, and everything seems to be working well now, with the caveat that there are still some issues with rotating through certain orientations of the end effector. 

The problem lies in the fact that certain axis aligned quaternions at some rotations around the aligned axis can have multiple equivalent values; when the IK algorithm tries to rotate through these orientation spaces, it is attempting to search for the quaternion with signs opposite the quaternion that is returned as its current position (for example, Q[0.707, -0.707i, 0j, 0k] == Q[-0.707, 0.707i, 0j, 0k]), and so is unable to recognize that it has, in fact, found the correct orientation for the end effector. For now, I have solved this problem by simply having the IK algorithm search for one of the two possible quaternions first, followed by the second if that proves unsuccessful. This works fairly well, but I'll be looking into alternative solutions to try and iron out some of the erratic behavior produced by the current solution. Still, with the last few changes implemented, rotational IK is starting to look very good, and now all that should be left is figuring out a method of quaternion calculation that produces quaternions consistent with the quaternion rotational transformation code.

neoaero99, 2016-06-06:
Hooray!

vtdrusch, 2016-06-10:
IK algorithms working as intended (to the best of my knowledge): rotation and translation are both functional, adhere to all specified joint angle limitations, and scale with the current speed of the robot. There are a couple of tweaks that could be made related to this issue, but as the core functionality of the IK algorithm is now complete, I think we can finally call this feature finished.

closed 2016-06-10 by vtdrusch