skuhl/RobotRun #11: Robotic arm should be able to pick up a cube

skuhl, 2016-05-04:
A simple implementation at first should be fine. I.e., is the claw at the cube and is it closed? If so, then the cube is picked up. Alternatively, we could explore options to add existing physics engines to the software. I think we may be able to avoid using a full-blown physics simulator.

neoaero99, 2016-05-19:
How should the orientation of an object in the world be handled? Should I store each object's orientation individually?

neoaero99, 2016-05-31:
So, you can pick up objects with the Robot arm now. I have fixed any bugs that I could think of for holding an object, moving it around and placing it again (though collision handling does not exist yet, so there my be problems later on). However, without an end effector offset, it is rather akward to pick up an object, because the end effector position must be touching the object's hit box in order to pick the object up.

neoaero99, 2016-06-08:
Yeah, this is completed as well ...

closed 2016-06-08 by neoaero99