skuhl/RobotRun #8: Figure out how to load a simple CAD model

skuhl, 2016-05-04:


vtdrusch, 2016-05-04:
I haven't examined all the details of the model loading process yet, but from a cursory examination of the system, it looks like model loading should be fairly straightforward for any STL file. I'm going to experiment with an STL model of a cube I found online.

vtdrusch, 2016-05-04:
I've managed to successfully load a cube model into the program environment. The cube currently is not animated and doesn't interact with the robotic arm at all, but it is apparently loaded and rendered correctly. The next major hurdle here will be implementing some kind of physics properties on these models.

closed 2016-05-04 by vtdrusch

skuhl, 2016-05-05:
Great! You may also want to confirm that we can get the files that are going to be created for us (or maybe by us?) into this format. Another thing to watch out for is that the units are correct (i.e., it is scaled to the right size and/or uses the right units) and that the origin of the model is where we expect it to be.