skuhl/RobotRun #12: Application not closing properly

vtdrusch, 2016-05-05:
The application throws some kind of error when closed using the window close ("X") button. The problem seems to be related to some rendering thread or process not being shut down properly when closing the application this way. For some reason, this method of exiting the application does not properly dispose of all components being used. 

I've found that the problem is not present in the older 2.x versions of processing, so it may be a good idea to discuss whether we should continue to use Processing3 or switch to the older version that this application was likely originally designed in moving forward. There seems to be no issue with using Processing 2.2.1 to run the application, but I'm not aware of all the differences between these two releases.

skuhl, 2016-05-05:
Are there any reports from other people online having the same problem? Do other (simple) Processing3 applications have this problem?

vtdrusch, 2016-05-05:
I've been researching the issue and haven't managed to track down anything
particularly illuminating, but I'll continue looking into it for the time
being. So far, it appears that closing the program window specifically by
pressing the window exit button doesn't call the correct disposal methods,
while using the escape key seems to call these methods and does close the
application properly.

skuhl, 2016-05-06:
Does calling exit() or similar from the code cause it to exit successfully? If so, then maybe you can hook onto the window handler, have it call your code, and call exit() in your code? Or, if you really get stuck, it might be worth asking a question on an appropriate forum or something.

vtdrusch, 2016-05-06:
The exit() function does correctly close the application, and much of my
efforts have been focused on trying to figure out how to capture the frame
event generated by pressing the close button so that I can include this
call. Unfortunately, it isn't entirely clear how to do this in Processing3.

As best I can tell, the "main" class, RobotRun, is an extension of the
class PApplet, and while it used to be the case that PApplet was itself an
extension of the Java Applet class, this has been changed in version 3. If
PApplets still run in a Frame object, I haven't been able to access it (the
frame field in PApplets for Processing3 just returns an integer) and I
haven't been able to determine how to add a PApplet to a Frame manually.

The applet does actually call a dispose method on close that can be
overridden, but the timeout error occurs before disposal.

I'm going to keep looking though, I'm confident that there is some way to
either intercept the window event or remove the offending function or
object that is causing the issue.

neoaero99, 2016-05-09:
You fixed this, right Vincent?

vtdrusch, 2016-05-09:
Yep, this issue should be rectified, I'll go ahead and close it.

closed 2016-05-09 by vtdrusch