skuhl/RobotRun #17: Limit camera movements

skuhl, 2016-05-08:
We need to limit the camera movements. Maybe always point the camera at the robot, and always have it be a certain distance from the robot---and let the camera rotate somewhere on a sphere.

skuhl, 2016-05-08:
If we implemented this, we probably do not need rotate button and a pan button.

vtdrusch, 2016-05-19:
I've made a couple of adjustments in this vein: first, rotation and panning mode are now mutually exclusive, so users will no longer have to contend with having the camera alternate between these two movement types with every mouse click. Selecting the rotate button on the UI (or pressing the corresponding keyboard key) will turn off panning if it is currently enabled, and vice versa.

Second, I've added mouse drag controls to allow the user to manipulate the camera. Pressing and holding the right mouse button while moving the mouse will allow the user to rotate the camera, and moving the mouse with both buttons held down will pan. Holding down both mouse buttons also disables camera rotation, so the camera will only perform one type of movement at a time.

These features should alleviate some of the frustration of having to contend with awkward camera movements.

neoaero99, 2016-08-24:
There are limitations for the panning of the camera now.

closed 2016-08-24 by neoaero99