skuhl/RobotRun #27: Make text highlighting more obvious

skuhl, 2016-05-08:
Right now we highlight text in red. We should maybe highlight the line and then highlight different part of the instructions (if appropriate).

neoaero99, 2016-05-17:
Despite there being setColorBackground() and setColorForeground() for a textLabel object in the ControlP5 library, neither of those methods seem to change the color of the label. So, the best I can do to highlight a line is highlight all of the test a specific color; would that be alright?

skuhl, 2016-05-17:
For now, anything that is a step in the right direction is good. We might have to come back and revisit all of the text to make it more legible in general though.

neoaero99, 2016-06-27:
Vincent has updated the UI to add more obvious highlighting.

closed 2016-06-27 by neoaero99