skuhl/RobotRun #31: Fix circular joint movement

vtdrusch, 2016-05-14:
The "C" type movement instructions are currently the only instruction type that doesn't seem to function at all. I've fixed a few crashes caused by trying to execute these instructions, but right now they don't appear to do anything other than halt program execution.

neoaero99, 2016-05-16:
Alright; do you want to reformat the program code first or continue working
on the circular motion?

vtdrusch, 2016-05-16:
Go ahead and keep doing what you've been doing, I'm going to look at a new
design for the program execution code later today.

neoaero99, 2016-05-16:
I realized that we need the circular motion to work in order to allow for
the rotational jogging in the World Frame.

vtdrusch, 2016-06-10:
Circular IK is complete, closing this issue (will still need to update C type instructions, but this is a different problem).

closed 2016-06-10 by vtdrusch