skuhl/RobotRun #32: Convert RobotRun to Eclipse compatible project

vtdrusch, 2016-05-16:
I had mentioned this before, but it may be easier to develop this application if we can use a more fully featured and stable IDE such as Eclipse. There are a couple of extensions available for Eclipse that includes this functionality, but I have been experimenting with the 'Proclipsing' extension to include a Processing based application in Eclipse.

It is relatively straightforward to get the project working in Eclipse once the extension is installed; simply adding the .java file generated when exporting the project in the Processing IDE to an Eclipse project allows the execution of the project with little hassle, so it is very doable to make this change. The Proclipsing extension also allows for the easy exporting of the project on multiple platforms (although it produces a batch script rather than an executable script, which may be less desirable). The quick and easy conversion process and the somewhat better featured IDE may make Proclipsing a good option to switch to moving forward, but I wanted to see what everyone else's thoughts were on the topic before doing any major work on an Eclipse based version of the project, as it may be less than trivial to return to the standard Processing project model in the future.

neoaero99, 2016-05-16:
I don't mind doing it

skuhl, 2016-05-16:
Is the "Proclipsing" extension something that is actively developed and
expected to be around for a while? How much more complex would it make the
our code after it is exported to Java? My general preference would be to
either use Java or use Processing---doing something that is in-between
doesn't seem like a great plan.

Regardless of which of the two languages we decide to use, you could use
whatever editor you wanted to edit the source code (of course).

While I can understand the shortcomings of the Processing editor. And, I
understand that this is the problem that we are trying to solve. But is the
proposed solution worth the extra complexity that might be involved with
the switch?

vtdrusch, 2016-05-16:
Whether or not the extension will continue to be supported in the future is a good question, and that's certainly a concern for future development, if not for the work we'll be doing this summer. All the extension really does is to automate the inclusion of the Processing code libraries, so it should be technically possible to simply treat the program as a Java project, although this would require some additional setup on the part of the programmer.

As far as code complexity, the source file and instructions are essentially identical to the Processing native files, but the whole project is in one big .java file rather than split among several Processing files, for better or worse (we could also unpack the nested classes used in the project to separate source files, although this may be time prohibitive). The only real difference from a programmer's perspective, once the project is set up in Eclipse, is the organization of the code into that single file.

That said, I'm not entirely sold on the switch myself, especially since it may be difficult to go back to Processing from Eclipse if we start making changes to the Java source file. I'm hesitant to make any major changes to the project structure too quickly, and I certainly don't want to be reliant on a tool that may not exist in a year's time.

vtdrusch, 2016-05-19:
Closing this issue for now, will continue to work with Processing native project structure.

closed 2016-05-19 by vtdrusch