skuhl/RobotRun #37: The Robot's Speed

neoaero99, 2016-06-10:
When I reviewed the 3-Point method with Siddharth, he mentioned that the Robot should be able to move at speeds slower than our current 1% speed.

vtdrusch, 2016-06-11:
I've reduced both the robot's translational and rotational base speed by 75%, which seems to have brought the robot's movement more in-line with that of the actual robotic arm's. Users should now be able to control the robot's position in a much more granular way, and we can also add a further step down from 1% speed for fine motor control.

vtdrusch, 2016-06-11:
Also tweaked the joint motor speed values for joint rotational movements. Interested to see how Alex and Siddharth like these changes.

neoaero99, 2016-06-11:
Cool, sounds great.

closed 2016-06-11 by vtdrusch