skuhl/RobotRun #40: Improve saving/ loading

vtdrusch, 2016-06-11:
Josh and I had already discussed this in part, but the original mechanisms for saving and loading programs and frames are somewhat inefficient and can lead to problems with the application. In particular, I noticed a lot of unchecked calls to Scanner.next() in the file load methods causing errors when the file the scanner is reading from happens to be empty. We should either provide more robust checks to avoid or handle these errors in an appropriate manor, or design the saving methods such that empty files aren't a concern.

neoaero99, 2016-06-21:
I probably should have mentioned that I have been working on converting saving to binary files, however, I have nearly completed the process.

neoaero99, 2016-06-21:
Saving has been fully converted to binary files for programs and frames.

Though I should mention that since we will be adding more variations to program instructions in the near future that in order for a program to be saved it NEEDS to have a saving case in the saveInstruction() AND a loading case in loadInstruction() methods.

closed 2016-06-21 by neoaero99