skuhl/RobotRun #53: Simplify menu interactions when modifying frames/ registers

vtdrusch, 2016-07-05:
The user should be able to open the relevant editor menu simply by selecting a given frame or register and pressing 'enter.'

neoaero99, 2016-07-06:
Do you mean the transition from the TOOL_FRAME_NAV or USER_FRAME_NAV to FRAME_DETAIL mode?

vtdrusch, 2016-07-06:
Floating point registers, position registers, and both tool and user frames should be accessed by selecting the desired frame or register in the appropriate menu and pressing enter. This mostly applies to the editing of frames, where the process of accessing the editing methods is somewhat complicated.

vtdrusch, 2016-07-10:
Based on what Siddharth and I had discussed earlier this week, registers
(data and position) should be accessed by selecting the desired register
from the appropriate menu and pressing enter, which is largely how they
work now but with a function button replacing the enter key.

For frames, the user should be able to enter the edit menu more easily;
rather than using a series of function presses, we should try to come up
with a more streamlined way of accessing the frame set functions. I'll have
to take another look at how the real world pendant handles this
interaction, and I think the general menu flow (going from frame select to
frame details to frame edit) is good, we just may want to look into a more
user friendly way to navigate through these menus.

neoaero99, 2016-07-13:
I modified the Register and Frame menu transitions, let me know what you think.

neoaero99, 2016-07-21:
I am pretty sure that this is done.

closed 2016-07-21 by neoaero99