skuhl/RobotRun #54: Provide quick transitions between 3, 4, and 6 point frame edit modes

vtdrusch, 2016-07-05:
The user should have a quick and simple way to convert between any of the frame entry modes. For example, after the user has specified a 3-point frame, they should be able to simply add a fourth point to the frame as the origin offset while preserving the original 3-point frame specifier.

neoaero99, 2016-07-11:
I do not quite understand what you want, because the 3-Point Method of a Tool Frame and a User Frame are completely different, so it does not make sense to convert between the two.

neoaero99, 2016-07-11:
Also, there is no benefit to convert form the 6-Point Method to the 3-Point Method for the Tool Frame or from the 4-Point Method to the 3-Point Method for the User Frame; the 6-Point Method and 4-Point Method are build-offs of their respective 3-Point Methods. The only case in which this would be useful is from the 3-Point Method to the 6-Point Method for the Tool Frame, since teaching the TCP can be annoying.

spsid13, 2016-07-11:
for example if you have recorded the tool frame in 6 point method. and now
if i  change the method to 3 point method then the positions recorded for
approach point 1 2 3 are still available but origin X and Z are deleted.
similarly for user frame if i have recorded using 3 point method and now i
want to change to 4 point method then the three recorded points are carried
forward and i now need to record only the system origin... let me know if
this is clear.

neoaero99, 2016-07-11:
I moved this issue to a new one

closed 2016-07-11 by neoaero99