skuhl/RobotRun #55: Display joint-angle and Cartesian-coordinate position registers in same menu

vtdrusch, 2016-07-05:
There are currently two separate menus for displaying position registers either as joint-angle values or as Cartesian values (x, y, z, w, p, r). Users should access position registers via a single menu, and should have the option of switching between joint-angle and Cartesian point specifications once they have accessed a given position register for viewing/ editing.

neoaero99, 2016-07-11:
The VIEW_POS_REG_J and VIEW_POS_REG_C are essentially the same, except for when you view the value of a register.

neoaero99, 2016-07-20:
The Point and Cartesian Position Register menus exists as parallel displays, where only the functionality of saving points and editing the registers differs between the Point and Cartesian menus.

closed 2016-07-20 by neoaero99