skuhl/RobotRun #56: Fix floating point register input

vtdrusch, 2016-07-05:
The process of specifying a register value has been observed to be broken in certain circumstances, will have to look into this further.

neoaero99, 2016-07-06:
How so?

vtdrusch, 2016-07-09:
When editing a floating point register, a 0.0 will be pre-pended to the
value, making it impossible to create a value of 0.1 or greater.

neoaero99, 2016-07-09:
Ah, I see the misunderstanding. The backspace key functions for floating point input, so if you wish to remove the 0.0, then you can use the backspace key. I just adding the initialization, because I thought it would be useful. You can initialize it to an empty string if you wish.

neoaero99, 2016-07-09:
The backspace key works in almost all input fields, at least it should.

closed 2016-07-09 by neoaero99