skuhl/RobotRun #61: Fixing the UI

neoaero99, 2016-07-16:
Alright, so I have found way too many problems with the UI and I am going to try and fix them. I know the UI just went through an overhaul, but I need to go through and fix the menu transition again.

vtdrusch, 2016-07-16:
I'm still in the middle of the UI update, I just pushed the changes I had so far after merging last night. I'll be working on transitions all weekend, so if you don't want to deal with the bugginess for now, just revert to the previous commit. Otherwise, you can feel free to start implementing menu transitions for registers and frames; I already have program editing and navigation 90% done, the changes just aren't live yet.

vtdrusch, 2016-07-17:
The instruction menus are largely functional from what I can tell. There are probably a couple of small things that I've missed, but we can iron those out pretty easily as we go. 

Frame navigation and editing is working for the most part, although I have less experience with these systems and will have to have Josh tell me if I missed anything here; viewing, selecting, loading, and editing should all be working just as they did prior to the navigation changes.

Registers are currently being displayed correctly, but I haven't quite finished fixing the editing interface yet. That should be done either tonight or tomorrow morning. Overall, there are still a couple of areas where the changes have not been properly applied, but these should be in the minority, and the screen transition process should be much more painless to apply to new screens going forward.

vtdrusch, 2016-07-18:
Pushed a couple more changes; the direct entry method is no longer completely broken, but I wasn't able to quickly figure out how the user was intended to modify the frame values, so while you can view the frame properties, you can't currently change them. 

Data menus are now mostly working in that it's possible to set a value for both data and position registers. Modifying a data register once its set is a little sketchy, as I'm unsure of whether there is a register clear function, and the current value is cleared every time the user selects a given register to edit. Position registers have the same issue as direct entry frames: you can view the register properties, but you can't edit them. Setting a position register to the current end effector position is still working. I was also unable to get comments working just yet, but that should hopefully be a quick fix.

The only other issue with the new screen transition system is the fact that I haven't properly re-implemented value set/ reset statements for the row, column, and option select variables. I had initially had the screen transition function reset everything to 0 on every transition, but this seemed like undesirable behavior in some circumstances. I'm in the process of determining on a case-by-case basis which variables should be reset on screen transition, but this should be relatively easy to fix during testing.

neoaero99, 2016-07-18:
I think, I can solve some of those problems, but I believe that the updating of contents, options, opt_select, row_select, col_select, text_render_start need to be done not only on a case-by-case level, but also need to be split between intital screen transitions and updating of the current screen based on the current mode. Would it be too much trouble to create a google spreadsheet for these transitions? It will help with the process of fixing this.

vtdrusch, 2016-07-18:
Sure, that's a good idea. I'll go through and map out as much of the UI as
I can when I get in this morning.

neoaero99, 2016-07-18:
I added a google spreadsheet to the shared Google Drive folder Robotics Summer Project/2016_RobotRun_Docs/ .

neoaero99, 2016-07-21:
I am pretty sure that this is completed.

closed 2016-07-21 by neoaero99