skuhl/RobotRun #62: unable to run the software on library desktops

spsid13, 2016-07-18:
It shows as a white window and doesn't seem to work on library computers

neoaero99, 2016-07-20:
Hey Siddharth, did you try the build I sent you on Monday on the labs? I tried the 64-bit version and it worked fine. Let me know if you still have problems.

neoaero99, 2016-08-19:
Since we published this project, most Windows Machines will not let this program run without manually allowing access for each file in the project. So, it is wise to edit the properties of the archive file BEFORE extracting it. In the bottom of the window there should be a checkbox for unblocking the file: enable it. Afterwards, when you extract the files the program should run fine.

closed 2016-08-19 by neoaero99