skuhl/RobotRun #64: world frame tcp position

spsid13, 2016-07-18:
when the end effector is selected and the coordinate is tool frame the rotation is about faceplate center which is the tcp. It should be the same when in world frame as well. Until the tool frame is taught about the end effectors such as vacuum cups center it should consider the tcp to be faceplate..

neoaero99, 2016-07-18:
I added default Tool Frames for both End Effectors that are set when either one is selected and there is no active Tool Frame. Do you want me to remove it?

spsid13, 2016-07-18:
In the final software it should be removed. i assume you might now need it for faster testing. Just wanted to highlight it so that we don't forget.

neoaero99, 2016-07-18:
Alright, will do.

neoaero99, 2016-07-19:
I removed the default Tool Frames for the End Effectors.

closed 2016-07-19 by neoaero99