skuhl/RobotRun #93: changing views using 123456 is removed. can we give separate icons on the top do so?

spsid13, 2016-08-16:


closed 2016-08-16 by spsid13

neoaero99, 2016-08-16:
They were not removed, I simply added the caveat that you must hold down enter, while pressing 1, 2, 3, 4, 5, or 6 to do the camera transitions. I added this so that the camera does not spaze out when you type numbers in the fields.

neoaero99, 2016-08-16:
This is why you also have to hold down enter to change axes displays. Also, you have to hold the right mouse button in order to scroll now, because the camera will zoom when you scroll through the dropdown menus without this. Though, I have not updated the main branch since I added this feature.

spsid13, 2016-08-16:
figured out... thats why closed it :)

neoaero99, 2016-08-16:
Alright, I just wanted to make sure. I know I did not anounce these changes. They are only mentioned in the commits.