skuhl/RobotRun #100: changing the speed in the program changes the location of the point.

spsid13, 2016-08-21:
No matter what the speed of the robot. once the location of a point is recorded. it should reach that location with any speed.

neoaero99, 2017-05-01:
There is an issue with inverse kinematics: when the speed is too slow, sometimes the robot halts for a time before continuing to move.

neoaero99, 2017-06-16:
For both linear interpolation and joint interpolation, the margin of error has been reduced, since this issue was open. In addition, the speed of the robot's motion is taken into account for determining when the robot is close enough to its destination.

No matter what, there will always be a margin of error (particularly for linear interpolation), however, I am closing this issue for now.

closed 2017-06-16 by neoaero99