skuhl/RobotRun #118: issue with programming points wrt to user frame

spsid13, 2016-09-07:
When we record a point in the program by default the point is with respect to user frame. if there is not user frame then it assumes its coordinates with respect to world frame.

Now if I have created a user frame using 4 point method and recorded a point in the program.
next i go back to frames and record a new point for the origin in the user frame.
now when i go back and run the program, the robot should move to a point with respect to this new origin.

spsid13, 2016-09-07:
Also if i re record all the points of the user frame then the robot should move to a position wrt to this new frame created.

neoaero99, 2016-09-07:
What exaclty are you having trouble with?

neoaero99, 2016-10-13:
Points recorded in a program do change with respect of the User frame. Though, there are still issues with converting between frames: see #116

closed 2016-10-13 by neoaero99