skuhl/RobotRun #119: unable to enter the value on the right PR

spsid13, 2016-09-11:
[image]

vtdrusch, 2016-09-24:
Editing position registers as operands is now possible in register statements.

neoaero99, 2016-09-24:
Um, I found a problem ...

You cannot confirm your input for the position register index, when using the P[x, y] option.

neoaero99, 2016-09-24:
More specifically, when you edit the second index, it prompts for both indices and works fine. However, when you try and edit the first index (which includes the 'P[' prefix), then the ENTER button does nothing.

vtdrusch, 2016-09-24:
I'm not seeing this behavior on my end, make sure you pull from the dev branch and try again.

neoaero99, 2016-09-25:
I pulled and you still cannot edit the first column ('P[x ') for a PR[x, y] operand.

vtdrusch, 2016-09-25:
I'm closing this issue again as I am unable to reproduce this bug. Check your code branch and try deleting any program saves. If you still have this problem, let me know exactly what you did to cause this bug, including your current screen, previous screen, and the full register statement you were trying to edit.

closed 2016-09-25 by vtdrusch