skuhl/RobotRun #122: when we import objects let the input by user be only source and scale

spsid13, 2016-10-07:
The bounding box formed using length height and width should be automatically formed based from scale.

neoaero99, 2016-10-07:
I do not know a way to create a bounding box for complex objects. Since, the center point of imported objects is ambigous, I cannot determine the bounds of the box.

skuhl, 2016-10-07:
If an axis-aligned bounding box will suffice, all you need to do is iterate through all of the vertices and take note of the largest X component, smallest X component, largest Y component, smallest Y component, etc that you encounter. You will end up with 6 numbers which define the bounds of the box. You can average the pairs of numbers together to get the center of the box. But, I don't understand the exact problem you are discussing, so perhaps this won't work.

neoaero99, 2016-10-13:
So, you no longer input the bounding box dimensions for complex shape parts, however, there is an issue with drawing complex shapes. I still need to figure out how to draw the shape inside the bounding box. Which will be another issue.

closed 2016-10-13 by neoaero99