skuhl/RobotRun #125: register instructions

spsid13, 2016-10-11:
step 1  I record a new point L p[1] 100% fine
step 2 I want to edit the P[1] and change it to Pr[3]
When I select the global register it does nothing..

neoaero99, 2016-10-11:
Have you inputted a value for Pr[3]?

neoaero99, 2016-10-11:
In other words, is Pr[3] already defined?

spsid13, 2016-10-11:
Ohh no.

neoaero99, 2016-10-11:
You cannot specify a position register that is uninitialized.

spsid13, 2016-10-11:
I should be able to specify it but the program should give an error. Otherwise those registers become useless since they cannot be used as variables in register equations.

neoaero99, 2016-10-11:
You can specify empty registers in register statements, you simply cannot specify empty position registers for motion instructions.

neoaero99, 2016-10-16:
You can now specific uninitialized global registers for motion instructions.

closed 2016-10-16 by neoaero99