skuhl/RobotRun #180: Euler Angle Nonsense

neoaero99, 2017-05-26:
So, I noticed that specifically the euler angle set rx=45, ry=0, rz=-90 degrees does convert properly. It did before May 15th and I would suspect it worked before we changed our euler angle conversion method. Also, I am not sure what other orientations do not work properly.

neoaero99, 2017-07-20:
Most of the conversion issues I have found have to deal with the conversion between euler angles and quaternions, or more accurately the conversion between rotation matrices and quaternions, since we removed the direct conversion between euler angles and quaternions. I have not had any trouble with conversions between euler angles and rotation matrices.

spsid13, 2017-10-24:
The funniest issue title. hahaha Clear frustration over euler angles.