skuhl/RobotRun #183: The offset instruction doesn't seem to be working.

spsid13, 2017-05-27:


vtdrusch, 2017-05-29:
Currently, the process to enable offsets for motion instructions is to highlight the third column of a motion instruction (P[ / RP[), press F5 ([Reg]), and then press F5 again ([Offset]). The offset function should be working, but the process to enable it could use some simplification; we'll look into that.

vtdrusch, 2017-05-29:
I did have to make minor corrections to the joint-mode offset motion, as well as fix an issue where the user was unable to select the first offset register when adding an offset. I'll push these updates in a build tonight.

neoaero99, 2017-06-13:
Offset menu interactions now resemble that of a motion instructions position.

closed 2017-06-13 by neoaero99