skuhl/RobotRun #187: something wrong with position register again

spsid13, 2017-06-02:
If you go to data and then position register. and then save a position to that register.
Now there is EDIT option on the right most side of that line. when I go there and press enter
I should be able to see the values but it doesn't show any. Also, like i mentioned earlier we need an option to change the representation of that position register from joint to linear and vice versa.

neoaero99, 2017-06-02:
You can change the position register type with F3, when shift is off. Also, you must be in the second column of the pendant screen to edit the position register. The first column with take you to the comment edit.

neoaero99, 2017-06-02:
What happens when you press enter over the position register? Do you get a blank screen? Could you send me you position register file?

spsid13, 2017-06-02:
It shows this when i press enter. How do you want me to send you the
position register file?
[image: Inline image 2]

spsid13, 2017-06-02:
> 248-516-9390 <(248)%20516-9390>

Hey Josh..
so when i recorded the position to position register it was in world mode
and when I change the type to cartesian I can see the positions but when i
change it to joint I see all zero.

spsid13, 2017-06-02:
> Did you record the robot's default position?

ok pick up my call.. it will be much easier to solve the issue.

neoaero99, 2017-06-02:
So, I found some deeper issues with the position offset that need to be addressed. Namely, we CANNOT simply add both the joint angles and the position and orientation together. You can only do either or and use inverse kinematics or forward kinematics to find the other values.

Also, we do not actually save the offset value for motion instructions, so yeah that need to happen.

neoaero99, 2017-06-13:
In general, offsets have been fixed.

closed 2017-06-13 by neoaero99