skuhl/RobotRun #191: IO Register Issue

neoaero99, 2017-06-13:
The evaluation of IO register in if statements is non-functional. I tried a simple expression:

IO[4] = IO[3]

however, the if statement failed to evaluate.

vtdrusch, 2017-07-07:
The equality operator (==) is meant to test the equality of floating point operands only, and so using boolean IO operands causes an evaluation error. I could extend the equality and inequality (!=) operators to test for boolean equality, but the preferred way of doing this currently is by using the logical AND (&&), OR (| |) and NOT (!) operators.

neoaero99, 2017-07-09:
The expression for my example has a size of one. It does not use the equals operator. The equals is the one pertaining the result element

neoaero99, 2017-07-09:
Never mind that last comment.

neoaero99, 2017-07-09:
And I see what you mean. That makes sense.

closed 2017-07-09 by neoaero99