skuhl/RobotRun #203: Also how easy it would be for you to export programs into notepad?

spsid13, 2017-06-25:


neoaero99, 2017-06-25:
Exporting a program to a text file would not be hard. However, we need to know what format is required for the text (i.e. how should each instruciton be outputted). I am not sure about other file types.

spsid13, 2017-06-25:
The same as it looks on the screen..
Each line with serial number its like printing the program

vtdrusch, 2017-07-06:
I've implemented a function that will automatically write programs to text every time the program is saved. Each program is saved to a file called <program name>.txt in the same folder as the program save file ("tmp/robot#"). You should see this functionality in our next release; feel free to let us know if you need any adjustments made to this feature.

closed 2017-07-06 by vtdrusch