skuhl/RobotRun #204: Delete is still not friendly

spsid13, 2017-06-25:
So we should also have something talking about deleting multiple lines press shift + enter
Now here's the thing If I want to delete lines 1-5
I press enter while the cursor is on line 1
then I go down to line 5 and now press Shift and enter
It does not delete all 5 lines. Not sure why.

neoaero99, 2017-06-27:
So, I want to get this whole shift select thing straight.

If you press shift and enter on a line select menu, then all elements between the last element you selected and the current element you selected should be selected/unselected.

Is that correct?

neoaero99, 2017-06-27:
I have added in a shift + enter functionality for line select menus like the delete option menu (for some reason the origin shift + enter functionality that Vincent made has been lost).

How it works is that, when you press enter, while shift is on, all lines from the last line you selected to the current line you selected are selected/unselected.

neoaero99, 2017-06-27:
Although, this functionality is not on the master branch, so until Vincent and I have tested the screens more on our developement branch, you will not see this change in a release.

vtdrusch, 2017-06-30:
I have fixed my implementation of multiple line selection with the shift key, which you will find in the most recent build. Let me know if there are any changes or improvements that need to be made to this feature.

closed 2017-07-07 by vtdrusch