skuhl/RobotRun #218: the trace mark erase option

spsid13, 2017-07-07:
So the the gluing I/O for example makes a trace when the program is run.. but now I want erase the trace and run the program again.

neoaero99, 2017-07-07:
If you disable the trace with the button in the miscellaneous menu, then
the trace will be cleared.

spsid13, 2017-07-07:
No its not working..

vtdrusch, 2017-07-07:
> 248-516-9390 <(248)%20516-9390>

I'll take a look at this and let you know when I have a new release.

spsid13, 2017-07-07:
> 248-516-9390 <(248)%20516-9390> <(248)%20516-9390>

Thanks Vince.

vtdrusch, 2017-07-07:
Could you describe how exactly you found this bug? To test the trace function, I just made a short program and enabled tracing through the misc. menu, but I haven't been able to reproduce the problem yet on the most recent build.

spsid13, 2017-07-07:
so turn on the I/O for gluing make the robot move and turn the I/O off..
when you run this program it will leave a trace for the robots movements.
Now there isn't a way to get rid of that trace.

spsid13, 2017-07-07:
by turning on IO i mean write a program that says
IO[4] = ON
p1
p2
p3
IO[4] = Off

vtdrusch, 2017-07-07:
I managed to clear the trace after running this program by clicking on the "Enable Trace" button in the miscellaneous menu, then pressing the same button again once the trace is enabled (it should then read "Disable Trace"). If this does not clear the trace for you, could you look in your I/O register menu and tell me if your I/O register was correctly set to OFF? Pressing the "Disable Trace" button should clear the trace, from the screen, but only if the I/O register associated with your current end effector is set to off as well (although I agree this is somewhat confusing, and I plan on adding a button to explicitly clear the trace, I just want to make sure there aren't any other problems preventing you from clearing the trace with the current version of the software).

spsid13, 2017-07-07:
It worked with the I/O off. Also can you assign the F key for turning IO on
and off in the I/O menu.

Yaa, it would be great if you could add a clear trace button that is
independent of I/o being on/off. Please let me know when you update and
push a build with these two changes.

vtdrusch, 2017-07-07:
I've implemented these features and will be uploading a new build with these changes in about an hour.

Turning the trace function on and off no longer clears the current trace, this must now be done explicitly by pressing the "Clear Trace" button that I've included in the miscellaneous menu. Pressing CTRL+F on your keyboard will now also toggle the state of the I/O register corresponding to your currently selected end effector. 

Unfortunately, while these changes should allow the user to create multiple disjoint trace lines in theory, the way that we currently store and render the trace line means that any trace segments created by repeatedly toggling the trace function on and off will still be connected. I plan to fix this in a future update, but it might take some time to implement.

closed 2017-07-07 by vtdrusch

neoaero99, 2017-07-10:
I probably should have mentioned this earlier, but Ctrl+P will toggle the state of the active end effector of the active robot. It has been implemented for some time.