skuhl/RobotRun #226: I/O menu

spsid13, 2017-07-16:
1) Hello, I think I mentioned this but in the menu shown in the image use the F keys to toggle the I/O. by F key I mean F1 F2 F3 F4 F5 you see on teach pendant. The menu looks very unfriendly as shown in the image. assign ON OFF to the F keys.
2) Let it be shown on the right if the I/O is on or off. where other things such as scenario coordinates frame etc. are present 
[image]

neoaero99, 2017-07-16:
Are you sure that information should be display on the righthand UI? That UI is already cluttered as is, adding anything else would be pushing it.

Also, what is wrong with this menu? It is really not that different from the other register menus.

Also, when are the I/O registers supposed to be bond to the function keys? You know that those are used by most menus, right?

spsid13, 2017-07-17:
Currently the I/O can be toggled ON and OFF using the ENTER key..
It is not intuitive.. if you assign ON OFF to the F keys it will be visible
and intuitive

neoaero99, 2017-07-18:
Ah, I thought you meant to bind each I/O register state to a function key, no that is easy enough.

neoaero99, 2017-07-18:
The I/O menu has been updated on the branch, current.

closed 2017-07-18 by neoaero99