skuhl/RobotRun #231: Blank Instruction Editing

neoaero99, 2017-07-28:
For the pendant program instruction navigation screen, editing a blank instruction (F4 on the teach pendant) should function similar to overwriting the instruction (F2 on the teach pendant).

vtdrusch, 2017-08-29:
I've added a case in the edit screen logic that transitions to the select instruction screen if the currently selected instruction does not match any of the instruction subtypes that we check for when editing an instruction, but this could potentially cause unintended behavior if we don't check for every subclass of Instruction when editing an instruction in the program.

neoaero99, 2017-08-30:
I have a solution for that: We can make the Instruction class abstract add a BlankInstruction class, which functions as our current Instruction class does, so we should not have any inheritance problems.

neoaero99, 2017-08-30:
I have implemented the solution specified above, so we should not run into any issues with new instructions and editing.

closed 2017-08-30 by neoaero99