skuhl/RobotRun #232: Expression Evaluation Issue

neoaero99, 2017-07-28:
Once again, there are issues with expression evaluation, which I have stumbled upon, when testing the NOT operator.

vtdrusch, 2017-07-31:
I have re-worked expression evaluation to finally correctly deal with the NOT operator, as well as fixed an issue with operator options being incorrectly listed for two adjacent operators.

neoaero99, 2017-07-31:
Alright, I probably should have updated this on Friday, but I believe I had already fixed expression evaluation, however, I wanted to test it further before finalizing this issue. In addition, the issue with expression evaluation is deeply rooted in some recent changes that you made Vincent, we can talk about it when you get in today.

vtdrusch, 2017-08-02:
Expression evaluation has been re-worked again and should now work much more reliably.

closed 2017-08-02 by vtdrusch