skuhl/RobotRun #233: Sub Expression Editing

neoaero99, 2017-07-28:
There is an issue with how the options for operators is determined for sub expressions.

For example, I have a sub expression as the first element of a register expression. The first element in the sub expression is a data register operand, however, when I try to add an operator to the right of the data register operand, the only options I have are the logic operators (AND, OR, NOT).

vtdrusch, 2017-07-31:
Thus far, I haven't been able to reproduce this behavior. The only situation in which it should be possible to see just the logic operators is in the case where the previous operand or sub-expression evaluates to a boolean type operand, and only for IO type register expressions and IF statements. I may have done something to fix this inadvertently while probing the NOT operator issue, but I'll leave this open for now until we can test with your specific program to verify whether or not this has been fixed.

vtdrusch, 2017-07-31:
Also, there seems to be another issue with editing sub expressions related to the opEdit variable being incorrectly set, causing the wrong operand to be edited in some cases, or resulting in a null pointer error.

vtdrusch, 2017-08-02:
Expression editing appears to be working again, closing this issue for now.

closed 2017-08-02 by vtdrusch