skuhl/RobotRun #236: Option to create multiple parts and fixtures

spsid13, 2017-08-13:
Someway to enter multiple parts or multiple fixtures.

neoaero99, 2017-08-13:
Of the same kind?

spsid13, 2017-08-13:
Yes

neoaero99, 2017-08-14:
So, if I include an input field for quantity, then that would suffice?

spsid13, 2017-08-14:
Yes, It should.

neoaero99, 2017-08-14:
It is worth mentioning, that you can create multiple of the same object by clicking create repeatedly. I know can be tedious, however, that would only be the case for large volumes of parts or fixtures. So, I am just curious why you want to add large quantities of the same object.

spsid13, 2017-08-14:
Just trying to give an option to the user to mention the quantity or any
other way to copy/paste a part or fixture created.

neoaero99, 2017-08-14:
I can add copy and pasting for world objects.

neoaero99, 2017-08-14:
I am worried about adding in world objects in bulk, because complex objects in particular take up a lot of memory. I was able cause memory errors with loading in less than a hundred grinders, so giving users a faster method of adding in a lot of world objects may not be wise.

spsid13, 2017-08-14:
That should do I suppose.

neoaero99, 2017-08-19:
I added the move and copy function to the edit window (also I grouped the delete functio with these funcitons) on the branches limbo and master (0583ead)

closed 2017-08-19 by neoaero99