skuhl/RobotRun #238: By default there is some part or fixture in the menu with the name WO.

spsid13, 2017-08-13:
Can we remove it?

neoaero99, 2017-08-13:
What do you mean? You can have empty scenarios.

spsid13, 2017-08-13:
If I make any changes to the position values in the image below nothing on the screen changes because there is no such part in the environment. It is there in the software by default but does not physically exist in the environment. Kinda confusing for users.
[image]

neoaero99, 2017-08-13:
That is not a part, by default if no element is selected in a dropdown
list, then the name od the dropdown list is displayed. That is name of that
dropdown list

neoaero99, 2017-08-14:
Though, I can add messages for trying to confirm editing when no world object exists.

neoaero99, 2017-08-19:
Now the edit menu will hide all editing options, when no objcet is selected on both limbo and master branches (a704bee)

closed 2017-08-19 by neoaero99