skuhl/RobotRun #246: Has a button been implemented to move all objects to their default values for a scenario?

spsid13, 2017-08-20:


neoaero99, 2017-08-20:
Yes, it is in the positon submenu of the edit menu

spsid13, 2017-08-20:
That is for an individual object.
In the scenario tab, give an option that moves all objects at once to their default.

neoaero99, 2017-08-20:
There are two buttons. One for individual objects and one is for all
objects.

neoaero99, 2017-08-20:
Restore Defaults functions for all objects.

spsid13, 2017-08-20:
We should move the restore default to the Miscellaneous tab.

neoaero99, 2017-08-20:
> 248-516-9390 <(248)%20516-9390>

Why? Shouldn't it be with all the other UI elements that manipulate an
objects position and orientation?

spsid13, 2017-08-20:
> issuecomment-323556676
> 248-516-9390 <(248)%20516-9390> <(248)%20516-9390>

I agree it should, but it doesn't seem intuitive. I suggest if we are
making overall changes to the scenario, we should use miscellaneous tab for
it. It will be very easy for users to use our default scenarios provided
with the software. If you put into EDIT tab there are a lot of clicks and
confusion.

neoaero99, 2017-08-20:
I am not sure what you mean, but the difference in clicks to get from any menu to the edit-edit sub menu and the miscellaneous menu is at most one.

spsid13, 2017-08-20:
I am Sorry. you are right. I assumed we have to select an object to see
that option.

neoaero99, 2017-08-20:
Don't worry about it. It is an odd ball button in that menu as it stands.

closed 2017-08-20 by neoaero99