skuhl/RobotRun #250: PASTE does not work as per the given document

spsid13, 2017-08-20:
I checked for POS ID.
When you paste the lines, it is supposed to retain all the position numbers but it doesn't do so.
Please correct it and check all others for right functioning as per the document shared earlier.

neoaero99, 2017-08-29:
I fixed the Pos ID and Position paste options, though, I am not entirely sure how the Rm Pos ID differs from R Pos ID, so that option may not work correctly.

vtdrusch, 2017-08-31:
We're reviewing the RM Pos ID copy option, and just wanted some clarification: when pasting using this option, should IO instructions be re-ordered at all? For example, if I have a motion instruction and an IO instruction, like so: 

1)   L  P[1]  200mm/s  FINE
2)   IO[1] = ON
3)   J  P[2]  20%  FINE

Should this be pasted as:

4)   L P[2]  200mm/s  FINE
5)   IO[1] = ON
6)   J  P[1]  20%  FINE

or as:

4)   L  P[2]  200mm/s  FINE
5)   J  P[1]  20%  FINE
6)   IO[1] = ON

The first example is how the paste function works now: motion instructions are modified to correctly reverse the robot's pattern of motion, but the IO instruction remains in the same order relative to the motion instructions. However, the second example makes it more convenient to reverse the actions of the robot when manipulating objects. If you wanted to pick up an object and then set it back down, the first example would have the IO instruction changing BEFORE the robot returned to the position where the object was picked up, but the second option would place the object back where it originally was once the IO instruction was appropriately modified. Let me know which you would prefer/ which of these is correct.

spsid13, 2017-10-24:
Hi Vince,
Sorry, seeing this now. Is it possible for you to check on the actual robot. I think it works like the first option but not very sure of it.