skuhl/RobotRun #251: What is the method to select multiple lines?

spsid13, 2017-08-20:
I was under the impression we had shift working to do that.
So I want to copy from line 2-6.
I press enter when on line 2. scroll down to line 6. press Shift and select enter. Then lines 2-6 get selected.

Has any other way been implemented?

neoaero99, 2017-08-20:
You must hold down shift to select multiple lines.

spsid13, 2017-08-20:
But pressing SHIFT selects the lines not under the users control. What if i
want to select 2-6 lines?

neoaero99, 2017-08-20:
> 248-516-9390 <(248)%20516-9390>

Select them one at a time

spsid13, 2017-08-20:
> 248-516-9390 <(248)%20516-9390> <(248)%20516-9390>

So, is it possible to do it the way I explained?

neoaero99, 2017-08-20:
> 248-516-9390 <(248)%20516-9390> <(248)%20516-9390> <(248)%20516-9390>
> auth/ANdpJiYUj9gwKVfHmNjG_

I will talk to Vincent, since he umplemented this.

vtdrusch, 2017-08-29:
I see what you mean about multiple line select; when I originally implemented this, I had thought you wanted to be able to select a page of lines at a time by pressing shift+up/down arrow, and this is how the feature was implemented, but this only allows you to toggle your selection of either one or 8 lines at a time. 

I've decided to remove this particular implementation, so pressing shift+up/down will no longer change the selection state of any lines, but pressing shift+enter will now work as you described, with all lines from your currently selected line to the line above/ below the previously selected line being toggled. We'll get this change in the next release for you to try, and you can let us know if you'd like any further adjustments to line selection.

closed 2017-08-29 by vtdrusch