skuhl/RobotRun #252: Lets Give an option to name/label the TOOL and USER Frames

spsid13, 2017-08-20:


neoaero99, 2017-08-20:
This already exists

spsid13, 2017-08-20:
Where?
I can't see it.

neoaero99, 2017-08-20:
You can give a frame a name when looking at a frames details.

spsid13, 2017-08-20:
So, the names should be visible on the main TOOL FRAME Screen. And after I have renamed it, where can I see the name?

neoaero99, 2017-08-20:
I dunno, you never specified what to do with the names, so they are
relatively useless right now.

spsid13, 2017-08-20:
Lol. I remember discussing that we should have the names on the front list for both user and tool frame menus. No point of having the name otherwise.

neoaero99, 2017-08-21:
So, you only want the names to visible for the frame navigation menus?

spsid13, 2017-08-24:
yes, the user should be able to address the frames by a name

neoaero99, 2017-08-29:
Frame names are shown in the frame navigation menus and the program instruciton navigation menu (8bfbf5b). These changes also exist in the 1.2.4 release.

closed 2017-08-29 by neoaero99