skuhl/RobotRun #253: Robot speed does not update realtime

spsid13, 2017-08-20:
In the earlier case, when the robot was in motion and user changed the speed while it's moving, the speed would update immediately and the robot moved with that speed. Now, the speed updates but the robot does not move with the updated speed.

neoaero99, 2017-08-20:
Do you mean for jogging? Or motion instructions?

spsid13, 2017-08-20:
So when the user is creating a tool frame and records a point. He then
presses move to. That is when it fails to move with updated speed while in
motion.

neoaero99, 2017-08-20:
> 248-516-9390 <(248)%20516-9390>

You mean for teaching, then?

neoaero99, 2017-08-20:
When teaching, if you specify the move to for a point, then you are
effectively running a joint motion instruciton. However, the speed of the
movement is defined when you begin the motion.

spsid13, 2017-08-20:
> 248-516-9390 <(248)%20516-9390> <(248)%20516-9390>

No, after it is taught.
So, record a point. Move the robot far away from that point. Press
Shift+Move to with the speed at 5%.
While it's moving, change the speed to 50%. The robot should move with 50%.
It doesn't.

neoaero99, 2017-08-29:
For frame point teaching methods and position register move to functions, the speed of the robot's motion can be changed, while the robot is moving (5e654351). These changes exist as of the 1.2.4 release.

closed 2017-08-29 by neoaero99