skuhl/RobotRun #255: Tool Frame 6 point method seems to have been broken.

spsid13, 2017-08-20:
Try to use 6 point method to create a tool frame, Record the Orient Origin, X direction, and Y Direction points. The coordinates created at the tool tip do not match these.

neoaero99, 2017-08-20:
What do you mean?

spsid13, 2017-08-20:
The coordinate axis that is visible showing the tool X direction, tool y direction does not match the directions taught.

spsid13, 2017-08-20:
[tmp.zip](https://github.com/skuhl/RobotRun/files/1237379/tmp.zip)

spsid13, 2017-08-20:
Use the temp folder and see the first tool frame created.

spsid13, 2017-08-20:
[image]

neoaero99, 2017-08-29:
The tool frame 6 point method has been modified to take the robot's current orientation into account when teaching the frame (a3200dc). These changes exist as of the 1.2.4 release.

closed 2017-08-29 by neoaero99