skuhl/RobotRun #256: Copied the folder from one computer to another and observing this error

spsid13, 2017-08-24:
[image]

neoaero99, 2017-08-24:
What OS does your computer have?

spsid13, 2017-08-24:
windows 10

neoaero99, 2017-08-24:
I would make sure your video graphics drivers for that computer are up to date.

neoaero99, 2017-08-29:
What type is your processor (32-bit or 64-bit)?
What key of graphics card do you have?

neoaero99, 2017-08-29:
I found an [issue](https://github.com/processing/processing/issues/4802) on Processing's Github page that may be relevant.

neoaero99, 2017-10-10:
This issue seems to be related to the graphics hardware or software. Here is a link to a thread describing a similar problem:
[https://intel-openport-v7.hosted.jivesoftware.com/thread/110921?start=30&tstart=0](https://intel-openport-v7.hosted.jivesoftware.com/thread/110921?start=30&tstart=0)

vtdrusch, 2017-11-27:
I'm going to close this issue, since I don't think there's anything that can be done on our end to fix it. Make sure your software/ drivers are up to date and if the problem persists, we can try to find a workaround.

closed 2017-11-27 by vtdrusch