skuhl/RobotRun #258: Can't get Robotrun Running

nick3085, 2017-10-19:
Hi Guys, 

I've downloaded the last version of robotRun, but can't get it started. 
The pdf guide says to run RobotRun.exe on windows and RobotRun on Linux/mac. 
On mac there starts a blank screen with nothing and on windows there is no .exe file. 
Can you help me out? 

I use jar launcher 15.0.1 on mac os High Sierra.
Also have a windows 10 machine 

I hope to hear from you 

Kind regards, 

Nick

neoaero99, 2017-10-19:
Did you unzip the file?

neoaero99, 2017-10-19:
Also, do you remember how long you waited for the blank screen? The program
takes a second or two to startup.

neoaero99, 2017-10-19:
As for the user guide, since we ported the project over to Eclipse, the executable is RobotRun.java regardless of what OS you are using, so the user guide is wrong. Thank you for pointing that out.

nick3085, 2017-10-19:
Thnx for the fast reply, yeah did unzip it.
The screen on mac stays blank for more than 15 min (started the RobotRun.java file)

➜  RobotRun java -jar RobotRun.jar
ControlP5 2.2.5 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
Missing data subfolder!
java.lang.NullPointerException
	at core.RobotRun.setup(RobotRun.java:3300)
	at processing.core.PApplet.handleDraw(PApplet.java:2393)
	at processing.opengl.PSurfaceJOGL$DrawListener.display(PSurfaceJOGL.java:907)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
	at jogamp.opengl.GLAutoDrawableBase$2.run(GLAutoDrawableBase.java:443)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:759)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:452)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)

neoaero99, 2017-10-19:
Ah, that is what I thought. I have updated the 1.2.5 release. If you download it again you should not get that error.

If you have any other problems look through the issues and comment or post a new issue, if you have any.

nick3085, 2017-10-19:
Thanks, downloaded the updated version from the master branch.
But there is no RobotRun.jar file anymore how can I start it?

neoaero99, 2017-10-19:
Did you clone the branch? You have to download the RobotRun package from
the releases on Github in order to get the jar file.

If you clone the master branch, then you will get the source code and you
would need the Eclipse IDE to run it in that case.

nick3085, 2017-10-19:
Yeah i did clone the brach, now downloaded the release and it works thnx!

closed 2017-10-19 by nick3085

neoaero99, 2017-10-19:
You are welcome.